arduino-libraries/Ethernet #201: Ethernet shield startup problem if no internet connection

mgnthe54, 2022-07-07:
Hello, I use Arduino Mega with w5100 and w5500 Ethernet shields. And the Ethernet library v2.0.
If the Ethernet cable (or internet does not work when booting the Arduino)
The Arduino reboots without stopping it remains blocked on the instruction: error= Ethernet.begin(mac); then reboot , it does not pass following the code:
if (error == 0) Ethernet. begin(mac, ip); // force the ip address if not acquired by box

Does version 2.0.1 fix this problem?
If not, how can this problem be corrected?
Thanks

mgnthe54, 2022-07-07:
Hello, to determine where it was not working I inserted different messages in different places of the program using Serial.println "messages";
Of course I also have this as parameters:

#define SECRET_MAC {0x90,0xA2,0xDA,0x0D,0xC1,0xC1} // here your MAC address...

byte mac[]=SECRET_MAC; // variable for your MAC address

IPAddress ip(192,168,1,145); // The default IP address that the shield will take if the box does not assign an IP.

If you reconnect the ethernet cable it works fine. But normally if the DHCP of the box does not respond the shield should initialize with IPAddress.

JAndrassy, 2022-07-07:
> If you reconnect the ethernet cable it works fine. But normally if the DHCP of the box does not respond the shield should initialize with IPAddress.

yes. and it does for me and for others a.f.a.i.k.
so what is different in your test?

mgnthe54, 2022-07-07:
I can send you if you wish the code of the sketch.
But I'm not the only one to encounter this problem, some suddenly apparently use the Ethernet2 library, but I send emails and the library that allows me to send emails (emailsender) is currently only compatible with the Ethernet! library.
I would have to try with an Arduino (normal not a mega) to see if the problem recurs too. But now I'm on vacation for 3 weeks...
Thanks