arduino-libraries/Ethernet #204: Not able to get the MAC ADDRESS from the shield unlike ESP 32

UdayChaduvula, 2022-09-09:
Hi I want to get the mac address from the ethernet like how we are able to get the mac address from the ESP 32. But for ethernet I am hardcoded the mac address in my code. I don't want to do that. I want to get the mac address from ethernet itself. Could you please let me know how can I get the mac address from ethernet

per1234, 2022-09-09:
Hi @UdayChaduvula . Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/c/using-arduino/networking-protocols-and-devices/27

closed 2022-09-09 by per1234

JAndrassy, 2022-09-09:
the Wiznet chips don't have the MAC address stored

UdayChaduvula, 2022-09-10:
Hi,

I have a doubt. How to reestablish the internet connection via ethernet
shield arduino?. When we unplug the ethernet cable or there is an issue in
internet connectivity ?