arduino-libraries/Ethernet #210: Problem setting IPAddress on Nano Every

robotsmith, 2022-12-08:
Hi,
FYI there is a problem of compatibility with the compiler 1.8.8 on megaAVR board. 
I use a W5500 chipset with a nano every, everything was working fine on the 1.8.6 compiler version (without any change of the code).
Unfortunately I coundn't set local IP with the new compiler version. localIP() showed me 0.0.0.0 instead of 192.168.1.11

Maybe this is absolutely not an issue from the Ethernet library itself. But you are noticed ;)
Best Regards!

per1234, 2022-12-14:
Hi @robotsmith. Thanks for your report.

>everything was working fine on the 1.8.6 compiler version

Please check to see whether or not it works with version 1.8.7 of the "**Arduino megaAVR Boards**" platform and then reply here to tell us what you discovered. That will narrow down the list of changes that might have caused the regression.

---

Please also provide a minimal sketch that demonstrates the problem.

gleanlux, 2023-03-06:
Hello!

Yesterday I was faced with the same error.
I analyzed it for a long time before I found this post.
I would like to summarize my experience, in case it would help to find the problem:

I use Arduino Nano Every + W5500.

Variations and their results:
1. Arduino megaAVR Boards 1.8.8 compiler, and Ethernet 2.0.1 library:
  - a, With all parameters specified: The ethernet module was unable to set a static IP address as described above. After initialization, only the configured DNS address could be retrieved, ip, gateway, mask was set to 0.0.0.0, no network connection could be established.
  - b, Entering mac and ip: same result as variation a.
  - c, By specifying mac only (dhcp): the ethernet module retrieved the ip address without any problem and network communication was possible!
****

2. Arduino megaAVR Boards 1.8.8 compiler, and Ethernet2 1.0.4 library.
I definitely wanted my program to work with the original Ethernet library, but I tried this variation as a test:
 -  The ethernet module worked correctly in both static and dhcp mode.
 ****

3. Arduino megaAVR Boards 1.8.7 compiler, and Ethernet 2.0.1 library.
After reading this post, I reverted to the original Ethernet library and downgraded the compiler:
  -  Incredibly, the ethernet module worked correctly, network communication was possible!

If I can be of any further help, please let me know!

SAURABHRAHATEKAR, 2023-05-20:
The latest versions are not working either. I tried what @gleanlux did and it worked.