arduino-libraries/Ethernet #237: How Does This Work (Ethernet.h)?

thestumbler, 2023-10-18:
There are comments in the file `Ethernet.h` saying that the code no longer uses the two include files `EthernetClient.h` and `EthernetServer.h`. That is true, but it does try to include two flies `Client.h` and `Server.h`, neither of which exist (their prepended-with-Ethernet cousins so exist, but seemingly aren't used).

Am I missing something?

per1234, 2023-10-18:
Hi @thestumbler. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

closed 2023-10-18 by per1234

thestumbler, 2023-10-18:
Let me rephrase it then, as it sure seems to be a bug. 

The main include file `Ethernet.h` tries to include two files which do not exist -- `Client.h` and `Server.h`.

per1234, 2023-10-18:
It is not a bug. As I told you already, post on the forum and we'll explain things to you.