arduino-libraries/Ethernet #239: Ethernet library doesn't work with DFR0850

coeurtech, 2023-10-20:
Arduino UNO v3 with DFRobot DFR0850 shield (w5500 chipset w/POE & SD card, https://wiki.dfrobot.com/Ethernet_and_PoE_shield_for_Arduino_W5500_SKU_DFR0850) using Ethernet library 2.0.2 immediately returns failure when Ethernet.begin is called and Ethernet.hardwareStatus returns EthernetNoHardware.

Shield fails to get IP address using DHCP when using Ethernet2 library BUT DOES work with static IP, so issue appears to be with the official Ethernet library.

Code being used:

#include <SPI.h>
#include <Ethernet.h>
static byte macAddr[] = { 0xDE, 0xFE, 0xA0, 0xE8, 0x13, 0x9E };
IPAddress myIP(192,168,1,10);
IPAddress myDNS(192,168,1,1);
IPAddress myGW(192,168,1,1);
IPAddress myMASK(255,255,255,0);

void setup() {
  Serial.begin(9600);
  delay(2500);
  Serial.print("\n");
  if (Ethernet.begin(macAddr) == 0) {
    Serial.println("Failed to obtaining an IP address using DHCP");
    if (Ethernet.hardwareStatus() == EthernetNoHardware) {
      Serial.println("Ethernet shield was not found.");
    } else {
      if (Ethernet.linkStatus() == 0) {
        Serial.println("Ethernet cable unplugged.");
      } else {
        Ethernet.begin(macAddr, myIP, myDNS, myGW, myMASK);
      }
    }
  }
}

void loop() {

}

M4rkWilhelm, 2023-12-20:
I also have a DFR8050 and can confirm that it also fails. I'm using the example hardware check sketch from this repo. I had hoped the issue was a bad board. Mine was purchased from Digikey last month.

coeurtech, 2024-01-04:
In testing it appears that the return values are moved over by 1 bit.  So when the result should be 0000 0001 what gets returned is 0000 0010.

PaulStoffregen, 2024-01-04:
Try editing w5100.h to change 14000000 at line 21 to a slower speed, like 4000000 or even 2000000 just for the sake of troubleshooting.

https://github.com/arduino-libraries/Ethernet/blob/master/src/utility/w5100.h#L21C4-L21C4

Maybe this particular shield has a slow 5V to 3V buffer chip and just can't handle the normal SPI clock speed?

If using Arduino IDE 2, your Ethernet library is probably installed in a hidden folder.  To discover its location, use File > Preference to turn on verbose output during compile.  Then in the messages about compile, scroll up and look for the info about which libraries were used.  You'll see the full pathname.  How to access hidden folders depends on your operating system, but that sort of info is easy to look up with Google search.

M4rkWilhelm, 2024-01-08:
I acquired a Leonardo board, uploaded the same sketch from my Uno boards, and the DFrobot shield works as intended. The sketch contained some pin mode setup code from the DFrobot wiki, which is apparently unnecessary as removing it did not stop the board from starting properly. No need to set an Ethernet.init pin either.

I will attempt Paul's suggested workaround for the Uno boards.

koenpunt, 2025-02-17:
@PaulStoffregen changing the `SPI_ETHERNET_SETTINGS` in the header file worked for me. Is there another way of changing that, without having to modify the header file?