arduino-libraries/Ethernet #250: server.available() doesn't work on Opta

fogzot, 2024-02-16:
The documentation for `server.available()` says:

> Gets a client that is connected to the server and has data available for reading. The connection persists when the returned client object goes out of scope; you can close it by calling client.stop().

But in the following code the connection is closed immediately at the end of the scope:
```
#include <Arduino.h>
#include <Ethernet.h>
#include <EthernetClient.h>
#include <EthernetServer.h>

IPAddress ip(192, 168, 3, 2);
IPAddress dns(192, 168, 3, 1);
IPAddress gateway(192, 168, 3, 1);
IPAddress subnet(255, 255, 255, 0);
byte MAC[] = {0xa8, 0x61, 0x0a, 0x50, 0x51, 0xff};

EthernetServer TelnetServer(23);

void setup()
{
    Serial.begin(9600);

    if (Ethernet.begin(MAC, ip, dns, gateway, subnet) != 0)
    {
        TelnetServer.begin();
    }
}

void loop()
{
    EthernetClient client = TelnetServer.available();
    if (client)
    {
        Serial.println("Client connected: " + String(client.remotePort()));
        int c;
        while ((c = client.read()) != -1) {
          client.write(c);  
        }
    }
}
```
The connection is supposed to persist across invocations of `loop()` but if one tries to connect (`telnet 192.168.3.2`) it is disconncted immediately.

JAndrassy, 2024-02-16:
yes. the networking libraries in mbed core implement `server.available()` as `server.accep()`.
if you need proper server.available() you can use ArduinoEthernetServer from my NetApiHelpers library. (works only with the latest mbed core release)

fogzot, 2024-02-16:
What about updating the documentation with this information? The docs sent me on a wild bug chase in *my* code before I guessed that there was an error in the library. And, IMHO, if the documentation says "A" and on every other platform it is "A", having it as "B" on mbed is a bug.

JAndrassy, 2024-02-16:
you guess were looking at the W5500 Ethernet library documentation, because the Mbed Core Ethernet library is not documented

fogzot, 2024-02-16:
Well, no. I am looking at the official documentation for the Ethernet library:

https://www.arduino.cc/reference/en/libraries/ethernet/

That reads:

> This library is compatible with all architectures so you should be able to use it on all the Arduino boards.

If a different, **undocumented** library exists and is used instead of the "official" one when i put `#include <Ethernet.h>` in my sketch, how am I supposed to know?

JAndrassy, 2024-02-16:
Opta doesn't use this library here.
it uses the bundled Ethernet library from Mbed Core
https://github.com/arduino/ArduinoCore-mbed/tree/main/libraries/Ethernet