arduino-libraries/Ethernet #251: Inconsistencies between Ethernet libraries of different architectures

soosp, 2024-02-19:
I'm working on a multi architecture project and foundthat there are inconsistencies of Ethernet libraries for different architectures. E.g.:
1. Ethernet library for AVR contains Dns.h (with more DNS related functions) while Ehternet for Opta doesn't
2. Ehternet for AVR doesn't contain hostname related functions (e.g.: Ethternet.setHostName())

It'd be good to synchronize different Arduino architectures at API level, to make it possible to create portable code.

soosp, 2024-02-19:
It was posted to https://github.com/arduino/ArduinoCore-mbed/issues/843 also.

JAndrassy, 2024-02-19:
https://github.com/JAndrassy/Arduino-Networking-API

soosp, 2024-02-19:
Thanks, it very informative. But I think, that the APIs (at least the ones maintained by Arduino) sould be in sync to support the cross-platform development.

JAndrassy, 2024-02-19:
>  But I think, that the APIs (at least the ones maintained by Arduino) sould be in sync to support the cross-platform development.

yes I am working on it. Ale the + in the tables are functions added since September23 and many PR are waiting to be merged