arduino-libraries/Ethernet #262: Stuck on parsePacket using the WIZ820io

jegjessing, 2024-05-03:
When trying to abtain a new lease, the module just hangs.
Tracing through the code I see that it gets stuck on int EthernetUDP::parsePacket()

Looking into the code, I see: // could this fail (loop endlessly) if _remaining > 0 and recv in read fails?

Answer is yes it could.
So now that I experience this. Is there anyone with a solution or some ideas as to how to help solve it?
(Mind you, I'm not the top 10 C developers in the workd :-) )

jslz25, 2025-10-24:
I fixed this in https://gitlab.mn.tu-dresden.de/josc941e/Ethernet/-/commit/137b823ed3458ed45ebb132813aa2b229fc2aec3, https://gitlab.mn.tu-dresden.de/josc941e/Ethernet/-/commit/ce865209be7adba4e57c5f778ce842f304883e64 and https://gitlab.mn.tu-dresden.de/josc941e/Ethernet/-/commit/a445484d01b39a1aabc699685ccc32b551f62690.  Could you try if you are satisfied with that change?

MrChencho, 2026-10-06:
Also Stuck on “parsePacket” using a W5500 along with an Arduino Nano. My application runs a program that makes one NTP request every hour and occasionally received UDP packets from remote sensors. Occasionally the program stops running (always on “parsePacket” after receiving an NTP answer, never processing a sensor UDP packet). The frequency of the fail varies from several times a day to one every few days (3 or 4).
After living with the problem for a couple of years (another processor in the system hardreset the nano when hunged), I compiled the program with a modified version of EthernetUdp.cpp (changed EthernetUDP::read functions) and it has been running for a couple of weeks without problems.
The change consists just in resetting _remaining (_remaining = 0) if there's no data available or recv failed.
Probably my solution is not the best one, so I´d appreciate if someone can develop a better one.

I can´t upload the modifies file, I paste below the changed code (lines commented in spanish)
-----------------------------------------------------------------------------
int EthernetUDP::read()
{
	uint8_t byte;

	if ((_remaining > 0) && (Ethernet.socketRecv(sockindex, &byte, 1) > 0)) {
		// We read things without any problems
		_remaining--;
		return byte;
	}

	// If we get here, there's no data available
	_remaining = 0; //si falla la lectura reseteamos _remaining para evitar bucle infinito en parsePacket
	return -1;
}

int EthernetUDP::read(unsigned char *buffer, size_t len)
{
	if (_remaining > 0) {
		int got;
		if (_remaining <= len) {
			// data should fit in the buffer
			got = Ethernet.socketRecv(sockindex, buffer, _remaining);
		} else {
			// too much data for the buffer,
			// grab as much as will fit
			got = Ethernet.socketRecv(sockindex, buffer, len);
		}
		if (got > 0) {
			_remaining -= got;
			//Serial.printf("UDP read %d\n", got);
			return got;
		}
	}
	// If we get here, there's no data available or recv failed
	_remaining = 0; //si falla la lectura reseteamos _remaining para evitar bucle infinito en parsePacket
	return -1;
}