arduino-libraries/Ethernet #266: Ethernet Problem

GabrielDSan, 2024-06-12:
Sorry to open this pull request, it has nothing to do with these files, my question is, I have artnet code along with ethernet on an esp32, but my current library doesn't work very well, only if I limit the computer port or the port switch to 10Mbps, at 100Mbps the library cannot handle packets correctly, I use the Webserver_esp32_W5500, I've been trying for more than months, if you could please leave some contact so I can describe my problem in full I would appreciate it, thank you in advance

per1234, 2024-06-12:
Hi @GabrielDSan. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

closed 2024-06-12 by per1234

JAndrassy, 2024-06-13:
try my new EthernetESP32 library with the WebServer library bundled with the esp32 platform