arduino-libraries/Ethernet #269: Subnet Configuration Problème

superkakaboum, 2024-06-29:
Hello,
I'm Gibril, a french student in networks and telecomunication, and we had to choose a subnet for our ethernet shield, but it didn't work (In the code (IPAddress subnet...) if I wanted it to be in /28 . I don't know why we need to change the W5100.cpp file, but also to change the subnet we need to go to the Ethernet .cpp file, line 75:

 void EthernetClass::begin(uint8_t *mac, IPAddress ip, IPAddress dns, IPAddress gateway)
{
	IPAddress subnet(255, 255, 255, 240); <--- Change this part
	begin(mac, ip, dns, gateway, subnet);
}

I'm not very good at programming and maybe I don't know how to use the arduino board but I'd like it to work in my arduino code when I set the subnet and whatever I want it to work so I don't have to go into the configuration files. 

For the record, I had to carry out a project to retrieve data from an angular sensor via a more or less complex network on gns3, a storage server which would then display the statistics on a web page.

JAndrassy, 2024-06-30:
there is `void begin(uint8_t *mac, IPAddress ip, IPAddress dns, IPAddress gateway, IPAddress subnet);` too

per1234, 2024-06-30:
Closing as invalid since, as **@JAndrassy** pointed out, it seems the library already provides the capability you require and the problem is simply that you aren't using the `EthernetClass::begin` overload that provides that capability.

@superkakaboum if you would like assistance with adjusting your sketch to use the correct `EthernetClass::begin` overload, or would like any other project discussion or assistance, you are welcome to post over on Arduino Forum:

https://forum.arduino.cc/c/using-arduino/networking-protocols-and-devices/27

I'm sure we'll be able to help you over there.

closed 2024-06-30 by per1234