arduino-libraries/Ethernet #271: EthernetUDP in Multicast - packet are sent to EthernetUDP source port instead of the destination port

NeyVian, 2024-07-11:
How to reproduce the issue :
```
EthernetUDP mcast_udp;
IPAddress mcast_group(230, 1, 1, 1);
uint16_t recv_port = 49500;
uint16_t dest_port = 49600;
mcast_udp.beginMulticast(mcast_group, recv_port);
mcast_udp.beginPacket(mcast_group, dest_port);
mcast_udp.write("some data");
mcast_udp.endPacket();
```

with this code the packet is sent from recv_port to recv_port, and not to dest_port.

This can be fixed in
socket.cpp > bool EthernetClass::socketSendUDP(uint8_t s)
by replacing 
`W5100.execCmdSn(s, Sock_SEND);`
with
`W5100.execCmdSn(s, ((W5100.readSnMR(s) & SnMR::MULTI) == SnMR::MULTI) ? Sock_SEND_MAC  : Sock_SEND);`

JAndrassy, 2024-07-12:
it is designed for use with the same port

NeyVian, 2024-07-12:
> it is designed to use the same port for receiving and sending

can you provide a source please ?

JAndrassy, 2024-07-12:
that is not a right use of SEND_MAC. from W5500 datasheet

> SEND_MAC transmits data without the automatic ARP-process. In this case, the destination hardware address is acquired from Sn_DHAR configured by host, instead of APR-process.

NeyVian, 2024-07-12:
> that is not a right use of SEND_MAC. from W5500 datasheet
> 
> > SEND_MAC transmits data without the automatic ARP-process. In this case, the destination hardware address is acquired from Sn_DHAR configured by host, instead of APR-process.

I'm not a specialist, but Multicast don't rely on ARP-process. no need to match an IP to a MAC as the group defines the destination MAC.
In my understanding, Multicasting is on Ethernet Level for hardware filtering not at UDP level.