arduino-libraries/Ethernet #274: EthernetServer ethServer(502) will not compile

jbr007, 2024-09-01:
Hi,

I am using ESP32-S3 and have some issue trying to setup the server.

#include <Arduino.h>
#include <SPI.h>
#include <Ethernet.h>

#include <ArduinoRS485.h> // ArduinoModbus depends on the ArduinoRS485 library
#include <ArduinoModbus.h>

byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };

EthernetServer ethServer(502);

will give error message "cannot declare variable 'ethServer' to be of abstract type 'EthernetServer'"

Any solution?

JAndrassy, 2024-09-01:
you use old version of the esp32 platform.
https://github.com/arduino-libraries/Ethernet/issues/241

per1234, 2024-09-01:
Closing as duplicate of https://github.com/arduino-libraries/Ethernet/issues/88, which has been resolved by https://github.com/espressif/arduino-esp32/pull/9447

@jbr007 you can fix the problem by updating to version 3.0.0 or newer of the "esp32" boards platform. Instructions here:

https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-arduino-ide

closed 2024-09-01 by per1234