arduino-libraries/Ethernet #277: Compilation error when using PlatformIO and Ethernet.h

MicSG-dev, 2024-12-19:
When using PlatformIO with the following platformio.ini file:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
    arduino-libraries/Ethernet@^2.0.2
```
And running the [WebServer](https://github.com/arduino-libraries/Ethernet/blob/master/examples/WebServer/WebServer.ino) example, several compilation errors occur:
```
src/main.cpp: In function 'void loop()':
src/main.cpp:75:43: error: cannot allocate an object of abstract type 'EthernetClient'
   75 |   EthernetClient client = server.available();
      |                           ~~~~~~~~~~~~~~~~^~
In file included from src/main.cpp:22:
.pio/libdeps/esp32dev/Ethernet/src/Ethernet.h:214:7: note:   because the following virtual functions are pure within 'EthernetClient':
  214 | class EthernetClient : public Client {
      |       ^~~~~~~~~~~~~~
In file included from C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32@src-e5a7fd5d9117bd8a322277fde59ac3d3/cores/esp32/Arduino.h:197,
                 from src/main.cpp:20:
C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32@src-e5a7fd5d9117bd8a322277fde59ac3d3/cores/esp32/Client.h:29:15: note:     'virtual int Client::connect(IPAddress, uint16_t, int32_t)'
   29 |   virtual int connect(IPAddress ip, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32@src-e5a7fd5d9117bd8a322277fde59ac3d3/cores/esp32/Client.h:31:15: note:     'virtual int Client::connect(const char*, uint16_t, int32_t)'
   31 |   virtual int connect(const char *host, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
src/main.cpp:75:18: error: cannot declare variable 'client' to be of abstract type 'EthernetClient'
   75 |   EthernetClient client = server.available();
      |                  ^~~~~~
Compiling .pio\build\esp32dev\FrameworkArduino\FunctionalInterrupt.cpp.o
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:40:16: error: invalid abstract return type for member function 'EthernetClient EthernetServer::available()'
   40 | EthernetClient EthernetServer::available()
      |                ^~~~~~~~~~~~~~
In file included from .pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:22:
.pio/libdeps/esp32dev/Ethernet/src/Ethernet.h:214:7: note:   because the following virtual functions are pure within 'EthernetClient':
  214 | class EthernetClient : public Client {
      |       ^~~~~~~~~~~~~~
In file included from C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32@src-e5a7fd5d9117bd8a322277fde59ac3d3/cores/esp32/Arduino.h:197,
                 from .pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:21:
C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32@src-e5a7fd5d9117bd8a322277fde59ac3d3/cores/esp32/Client.h:29:15: note:     'virtual int Client::connect(IPAddress, uint16_t, int32_t)'
   29 |   virtual int connect(IPAddress ip, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32@src-e5a7fd5d9117bd8a322277fde59ac3d3/cores/esp32/Client.h:31:15: note:     'virtual int Client::connect(const char*, uint16_t, int32_t)'
   31 |   virtual int connect(const char *host, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: In member function 'EthernetClient EthernetServer::available()':
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:47:54: error: invalid cast to abstract class type 'EthernetClient'
   47 |         if (!chip) return EthernetClient(MAX_SOCK_NUM);
      |                                                      ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:72:40: error: invalid cast to abstract class type 'EthernetClient'
   72 |         return EthernetClient(sockindex);
      |                                        ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: At global scope:
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:75:16: error: invalid abstract return type for member function 'EthernetClient EthernetServer::accept()'
   75 | EthernetClient EthernetServer::accept()
      |                ^~~~~~~~~~~~~~
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: In member function 'EthernetClient EthernetServer::accept()':
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:82:54: error: invalid cast to abstract class type 'EthernetClient'
   82 |         if (!chip) return EthernetClient(MAX_SOCK_NUM);
      |                                                      ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:104:40: error: invalid cast to abstract class type 'EthernetClient'
  104 |         return EthernetClient(sockindex);
      |                                        ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: In member function 'virtual size_t EthernetServer::write(const uint8_t*, size_t)':
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:170:18: error: cannot allocate an object of abstract type 'EthernetClient'
  170 |         available();
      |         ~~~~~~~~~^~
*** [.pio\build\esp32dev\libe72\Ethernet\EthernetServer.cpp.o] Error 1
```

per1234, 2024-12-19:
It is caused by the ESP32 platform deviating from the standardized Arduino Client class API:

https://github.com/espressif/arduino-esp32/issues/2755

They brought the platform back into compliance some time ago: https://github.com/espressif/arduino-esp32/pull/3191

But then there was a recent regression: https://github.com/espressif/arduino-esp32/pull/10202

I am not knowledgeable in the subject matter of PlatformIO, but my understanding is that the PlatformIO platform is significantly outdated: https://github.com/platformio/platform-espressif32/issues/1225, so I guess you are probably still suffering from the original incarnation of the bug rather than the current one.

MicSG-dev, 2024-12-19:
Got it. Thanks for the links. I was able to fix the errors temporarily by applying the changes in the Client.h file (removing line 29 https://github.com/espressif/arduino-esp32/blob/9eb7dc6f7bddf98600f6e8de3ccc19cc8ba5281b/cores/esp32/Client.h#L29 and line 31 https://github.com/espressif/arduino-esp32/blob/9eb7dc6f7bddf98600f6e8de3ccc19cc8ba5281b/cores/esp32/Client.h#L31) as per pull request 3191 https://github.com/espressif/arduino-esp32/pull/3191/files.

MicSG-dev, 2024-12-24:
Thank you very much for the information!! I will eagerly await the result.