arduino-libraries/Ethernet #280: spi0 and spi1

allanrozek, 2025-03-18:
Hello, I'm working on a C++ project with a Raspberry Pico 2, two W5500 Ethernet modules and a W2815 LED strip. The idea is to receive ArtNet data on a W5500 module in SPI0 (this part works), and to retransmit the same data on the second module in SPI1. The idea is to create a module chain. My problem is that I can't initialize the SPI1 bus via the CS pin (13 by default). Do you have any idea where the problem could come from? Thanks in advance.

JAndrassy, 2025-03-18:
use the Ethernet library from the RP2040 platform
https://github.com/earlephilhower/arduino-pico/blob/master/libraries/lwIP_Ethernet/examples/EthernetLibCompatible/EthernetLibCompatible.ino

allanrozek, 2025-04-09:
Hi, the problem is that the artnetcompat.h library is not compatible with the artnet.h library. Despite several attempts to modify lines in the header files, I continue to get compilation errors...