arduino-libraries/Ethernet #283: Weirdness setting second and fourth octet of ip address for w5500 module (must be <128)

PbZepplin, 2025-04-11:
Using Ethernet library v2.0.2, IDE 2.3.6, UNO with W5500 module
Starting from the webserver example, I'm having trouble setting an IP address with the second (bbb) and fourth (ddd) bytes (aaa.bbb.ccc.ddd) with a value greater than 127

For example, the output of ip(192, 128, 1, 1) yields a response from Ethernet.localIP() as 192.0.2.2

I've narrowed it down to anything greater than 127 makes an overflow into the next octet. A list of set IPs vs their outcome is;
`set to 	192.1.1.11
is 	192.1.1.11
set to 	192.10.1.1
is 	192.10.1.1
set to 	192.100.1.1
is 	192.100.1.1
set to 	192.200.1.1
is 	192.144.2.2   	<--bonkers
set to 	192.150.1.1	
is 	192.44.2.2	<--bonkers
set to 	192.127.1.1
is 	192.127.1.1 
set to 	192.128.1.1
is 	192.0.2.2	<--bonkers
set to 	192.127.127.1
is 	192.127.127.1 
set to 	192.127.128.1
is 	192.127.128.1
set to 	192.127.200.1
is 	192.127.200.1
set to 	192.127.200.127
is 	192.127.200.127
set to 	192.127.200.128
is 	192.127.200.0	<--bonkers`

So, I don't know what's special about those two positions in the IP, but I have no idea what's causing this... I've been able to recreate the issue with all of the provided examples (as they all use .168. in the second byte) from the library.
Any help appreciated!

PbZepplin, 2025-04-12:
P.s. I came across this dude also experienced the issue while troubleshooting, but no resolve given
[https://arduino.stackexchange.com/questions/92707/incorrect-ip-address-when-using-ethernet-library-with-arduino-uno-and-wiznet-550](url)

JAndrassy, 2025-04-12:
I can't reproduce the problem. show us the sketch which generated that output.

PbZepplin, 2025-04-12:
G'day Juraj,
Thanks for looking into it!
I made that table up by hand; changing the octet, uploading and Ctrl+C/+V the serial output into here. The code was the server example with only the ip variable changed. The sketch & hardware behaves itself otherwise. E.g.
- reports cable disconnected if absent
- if I set my PC to an IP in the same subnet as what the W5500 reports I can ping it (but the http:// page doesn't load)

`/*
  Web Server

 A simple web server that shows the value of the analog input pins.
 using an Arduino Wiznet Ethernet shield.

 Circuit:
 * Ethernet shield attached to pins 10, 11, 12, 13
 * Analog inputs attached to pins A0 through A5 (optional)

 created 18 Dec 2009
 by David A. Mellis
 modified 9 Apr 2012
 by Tom Igoe

 */

#include <SPI.h>
#include <Ethernet3.h>

// Enter a MAC address and IP address for your controller below.
// The IP address will be dependent on your local network:
byte mac[] = {
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED
};
IPAddress ip(192, 168, 1, 177);

// Initialize the Ethernet server library
// with the IP address and port you want to use
// (port 80 is default for HTTP):
EthernetServer server(80);

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

  // start the Ethernet connection and the server:
  Ethernet.begin(mac, ip);
  server.begin();
  Serial.print("server is at ");
  Serial.println(Ethernet.localIP());
}

void loop() {
  // listen for incoming clients
  EthernetClient client = server.available();
  if (client) {
    Serial.println("new client");
    // an http request ends with a blank line
    boolean currentLineIsBlank = true;
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        Serial.write(c);
        // if you've gotten to the end of the line (received a newline
        // character) and the line is blank, the http request has ended,
        // so you can send a reply
        if (c == '\n' && currentLineIsBlank) {
          // send a standard http response header
          client.println("HTTP/1.1 200 OK");
          client.println("Content-Type: text/html");
          client.println("Connection: close");  // the connection will be closed after completion of the response
          client.println("Refresh: 5");  // refresh the page automatically every 5 sec
          client.println();
          client.println("<!DOCTYPE HTML>");
          client.println("<html>");
          // output the value of each analog input pin
          for (int analogChannel = 0; analogChannel < 6; analogChannel++) {
            int sensorReading = analogRead(analogChannel);
            client.print("analog input ");
            client.print(analogChannel);
            client.print(" is ");
            client.print(sensorReading);
            client.println("<br />");
          }
          client.println("</html>");
          break;
        }
        if (c == '\n') {
          // you're starting a new line
          currentLineIsBlank = true;
        }
        else if (c != '\r') {
          // you've gotten a character on the current line
          currentLineIsBlank = false;
        }
      }
    }
    // give the web browser time to receive the data
    delay(1);
    // close the connection:
    client.stop();
    Serial.println("client disconnected");
  }
}

`

JAndrassy, 2025-04-13:
try to set the reset of the 'begin' parameters for static IP
https://docs.arduino.cc/libraries/ethernet/#%60Ethernet.begin()%60

PbZepplin, 2025-04-13:
Thanks for the review,
What do you mean set the reset? I've took another look at the .begin function with nothing obvious to me... I tried setting 
`byte ip[] = {192, 168, 0, 1);`
with the same result.
Also I tried the 
'Ethernet.setLocalIP(newIp);'
Function later in the setup() block with another ip to the same result. I'm beginning to wonder if this is a hardware fault in the module (which is confusing as the Ethernet.hardwareStatus() return is valid, as well as the Ethernet.linkStatus()
Cheers
Sam

JAndrassy, 2025-04-13:
a typo. "rest" no "reset"

PbZepplin, 2025-04-13:
Thanks for the suggestion. I fill out the other parameters of .begin() and also took the code snippets from the examples to print MAC, gateway etc also with some interesting results!!
Code:
```/*
 Web Server

 A simple web server that shows the value of the analog input pins.
 using an Arduino WIZnet Ethernet shield.

 Circuit:
 * Ethernet shield attached to pins 10, 11, 12, 13
 * Analog inputs attached to pins A0 through A5 (optional)

 created 18 Dec 2009
 by David A. Mellis
 modified 9 Apr 2012
 by Tom Igoe
 modified 02 Sept 2015
 by Arturo Guadalupi
 
 */

#include <SPI.h>
#include <Ethernet.h>

// Enter a MAC address and IP address for your controller below.
// The IP address will be dependent on your local network:
byte mac[] = {
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED
};
IPAddress ip(192, 168, 1, 177);
IPAddress dns(192,168,1,1);
IPAddress gateway(192,168,1,1);
IPAddress subnet(255,255,255,0);

// Initialize the Ethernet server library
// with the IP address and port you want to use
// (port 80 is default for HTTP):
EthernetServer server(80);

void setup() {
  // You can use Ethernet.init(pin) to configure the CS pin
  //Ethernet.init(10);  // Most Arduino shields
  //Ethernet.init(5);   // MKR ETH Shield
  //Ethernet.init(0);   // Teensy 2.0
  //Ethernet.init(20);  // Teensy++ 2.0
  //Ethernet.init(15);  // ESP8266 with Adafruit FeatherWing Ethernet
  //Ethernet.init(33);  // ESP32 with Adafruit FeatherWing Ethernet

  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  Serial.println("Ethernet WebServer Example");

  // start the Ethernet connection and the server:
  Ethernet.begin(mac, ip,dns,gateway,subnet);

  // Check for Ethernet hardware present
  if (Ethernet.hardwareStatus() == EthernetNoHardware) {
    Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
    while (true) {
      delay(1); // do nothing, no point running without Ethernet hardware
    }
  }
  if (Ethernet.linkStatus() == LinkOFF) {
    Serial.println("Ethernet cable is not connected.");
  }

  // start the server
  server.begin();
    Ethernet.begin(mac, ip);

  byte macBuffer[6];  // create a buffer to hold the MAC address
  Ethernet.MACAddress(macBuffer); // fill the buffer
  Serial.print("The MAC address is: ");
  for (byte octet = 0; octet < 6; octet++) {
    Serial.print(macBuffer[octet], HEX);
    if (octet < 5) {
      Serial.print('-');
    }
  }
  Serial.print("server is at ");
  Serial.println(Ethernet.localIP());
  Serial.print("The DNS server IP address is: ");
  Serial.println(Ethernet.dnsServerIP());
  Serial.print("The gateway IP address is: ");
  Serial.println(Ethernet.gatewayIP());  
  Serial.print("The subnet mask is: ");
  Serial.println(Ethernet.subnetMask());
}

void loop() {
  // listen for incoming clients
  EthernetClient client = server.available();
  if (client) {
    Serial.println("new client");
    // an HTTP request ends with a blank line
    bool currentLineIsBlank = true;
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        Serial.write(c);
        // if you've gotten to the end of the line (received a newline
        // character) and the line is blank, the HTTP request has ended,
        // so you can send a reply
        if (c == '\n' && currentLineIsBlank) {
          // send a standard HTTP response header
          client.println("HTTP/1.1 200 OK");
          client.println("Content-Type: text/html");
          client.println("Connection: close");  // the connection will be closed after completion of the response
          client.println("Refresh: 5");  // refresh the page automatically every 5 sec
          client.println();
          client.println("<!DOCTYPE HTML>");
          client.println("<html>");
          // output the value of each analog input pin
          for (int analogChannel = 0; analogChannel < 6; analogChannel++) {
            int sensorReading = analogRead(analogChannel);
            client.print("analog input ");
            client.print(analogChannel);
            client.print(" is ");
            client.print(sensorReading);
            client.println("<br />");
          }
          client.println("</html>");
          break;
        }
        if (c == '\n') {
          // you're starting a new line
          currentLineIsBlank = true;
        } else if (c != '\r') {
          // you've gotten a character on the current line
          currentLineIsBlank = false;
        }
      }
    }
    // give the web browser time to receive the data
    delay(1);
    // close the connection:
    client.stop();
    Serial.println("client disconnected");
  }
}
```

And serial output
19:34:11.835 -> The MAC address is: **DE-5B-7D-DF-FD-DB**server is at **192.80.3.98**
19:34:11.899 -> The DNS server IP address is: 192.168.1.1
19:34:11.978 -> The gateway IP address is: **192.80.2.3**
19:34:11.978 -> The subnet mask is: 255.255.255.0

So oddly enough, it's also the MAC and gateway which is being messed with!!!
Suffice to say, I'm even further from having a clue.
Cheers

JAndrassy, 2025-04-13:
it must be a hw problem. if you have SD card in the Ethernet shield, remove it

PbZepplin, 2025-04-13:
Nah, no SD card. Just a plain module like this;
[https://www.ebay.com.au/itm/224173348429?var=523162224418](url)
Thanks for the review, I'll buy a new module and give it a crack.

closed 2025-04-13 by PbZepplin